sparkfun/Qwiic_I2C_Py #2: FreeBSD support

tykling, 2021-01-29:
Hello,

I am playing around with a bunch of QWIIC sensors on a pizero running FreeBSD. I would _love_ to see support for FreeBSD in this so things are easier.

I can already speak to the QWIIC devices using the ```/dev/iic0``` device and ```i2c(8)``` but Python support would be fantastic.

I am available for testing stuff out or any questions!

Thanks :)

santaimpersonator, 2021-01-29:
I have transferred the issue over as it focuses on the implementation of the I2C communication drivers. Currently, we are focusing our efforts on getting more modules (devices) added to the `sparkfun-qwiic` package.

However, feel free to submit a pull request... obviously, don't forget to implement a way to recognize specific linux distributions as well.

tykling, 2021-01-31:
>  obviously, don't forget to implement a way to recognize specific linux distributions as well.

Hm? I don't use Linux. Any PR from me would focus on the topic of this issue, which is FreeBSD support. It is not obvious to me why I would include anything to recognize specific linux distributions? Maybe I misunderstood you :)

santaimpersonator, 2021-02-01:
I wasn't familiar with FreeBSD, so I had to google it and according to [Wikipedia: "FreeBSD is a free and open-source Unix-like operating system descended from the Berkeley Software Distribution, which was based on Research Unix..."](https://en.wikipedia.org/wiki/FreeBSD) As FreeBSD is a linux OS, in order to integrate support for FreeBSD into the [Qwiic_Py](https://github.com/sparkfun/Qwiic_Py) package, you would need to implement modifications within the [Qwiic_I2C_Py](https://github.com/sparkfun/Qwiic_I2C_Py) package, which provides _"multi-platform I2C bus integrations of the SparkFun qwiic ecosystem"_.

As far as I can tell, there are 3 ways for you to get support for FreeBSD:

1. A programming language other that Python.
    * In which case, I am closing this request as this is a Python package.
2. An I2C bus communication method separate from the `smbus2` Python package, used for our supported linux platforms.
    * In which case, you need to provide a method for the Qwiic_I2C_Py package to recognize FreeBSD from other linux distributions.
3. Provide support for the `smbus2` Python package on FreeBSD.
    * In which case, I am closing this request and you need to look into this repository: https://github.com/kplindegaard/smbus2/

tykling, 2021-02-01:
Hello :)

I agree with your three points, but I have to point out:

> As FreeBSD is a linux OS

No, it really, really isn't. FreeBSD has nothing to do with Linux. They do not share a kernel, userland, or anything like that. FreeBSD is not a Linux distribution, it is a completely seperate OS.

> As far as I can tell, there are 3 ways for you to get support for FreeBSD:
>
>     1. A programming language other that Python.
>
>        * In which case, I am closing this request as this is a [...]

2 or 3 will be the solution for this, but this issue will need to remain open, because changes in this package are needed too, once 2 or 3 is fixed (at least a new freebsd_i2c.py file for the bits speaking to the driver).

Differentiating between Linux and FreeBSD will be no different than differentiating between Linux and CircuitPython today, just an extra line here https://github.com/sparkfun/Qwiic_I2C_Py/blob/master/qwiic_i2c/__init__.py#L69 and we should be good.

santaimpersonator, 2021-02-01:
Semantics aside... yes, that was the gist of my point. Make sure you implement at way for the package to recognize FreeBSD separately from the currently supported platforms. (_*Sorry, I wasn't familiar with FreeBSD or where it lies in the linux/unix tree... like I said, I had to google it._)

I am hiding this comment and the last 3 comments, since they aren't productive to this issue. Also, I will correct my original comment to specify "platform" instead of linux distribution.

tykling, 2021-02-01:
:+1: will get back when I have gotten FreeBSD support in smbus2

santaimpersonator, 2025-12-12:
I'm going to close this issue, as there has been no updates for a while now.

Feel free to reopen the issue if you'd like. However, please keep in mind that this is beyond the scope of our support. Therefore, it would require development and maintenance by the community.

closed 2025-12-12 by santaimpersonator