sparkfun/Qwiic_I2C_Py #4: Camel Case Method Names

santaimpersonator, 2021-02-22:
Several of the method names are camel case, which isn't "pythonic"

[\_\_init\_\_.py](https://github.com/sparkfun/Qwiic_I2C_Py/master/qwiic_i2c/__init__.py):
- [getI2CDriver()](https://github.com/sparkfun/Qwiic_I2C_Py/blob/master/qwiic_i2c/__init__.py#L79)
- [isDeviceConnected()](https://github.com/sparkfun/Qwiic_I2C_Py/blob/master/qwiic_i2c/__init__.py#L116)

[circuitpy_i2c.py](https://github.com/sparkfun/Qwiic_I2C_Py/blob/master/qwiic_i2c/circuitpy_i2c.py), [linux_i2c.py](https://github.com/sparkfun/Qwiic_I2C_Py/blob/master/qwiic_i2c/linux_i2c.py), [i2c_driver.py](https://github.com/sparkfun/Qwiic_I2C_Py/blob/master/qwiic_i2c/i2c_driver.py):
- _connectToI2CBus():
- isPlatform()
- readWord()
- readByte()
- readBlock()
- writeCommand()
- writeWord()
- writeByte()
- writeBlock()

2bndy5, 2021-03-13:
[PEP8 standard has no problem with it](https://www.python.org/dev/peps/pep-0008/#naming-conventions). In my personal experience, the non-pythonic naming is usually due to a C extensions' naming convention (which follows popular C style naming). For examples, I defer to MicroPython/CircuitPython I2C APIs (& the CPython's 3<sup>rd</sup>-party SpiDev wrapper module also uses non-pythonic naming for the same reason).

santaimpersonator, 2021-03-14:
@2bndy5 Thanks for the information. We had a few customer comments on this topic, which is why I filed the issue. I think they were referring to these sections in that documentation:
- [Function and Variable Names](https://www.python.org/dev/peps/pep-0008/#function-and-variable-names)
  > Function names should be lowercase, with words separated by underscores as necessary to improve readability.
  > 
  > mixedCase is allowed only in contexts where that's already the prevailing style (e.g. threading.py), to retain backwards compatibility.
- [Method Names and Instance Variables](https://www.python.org/dev/peps/pep-0008/#method-names-and-instance-variables)
  > Use the function naming rules: lowercase with words separated by underscores as necessary to improve readability.

2bndy5, 2021-03-14:
Always good to document/discuss changes before committing to something. The only other concern worth discussing is backward compatibility. I see this repo is a submodule to [another Sparkfun I2C "unifying" lib](https://github.com/sparkfun/Qwiic_Py). I'm not too familiar with it (yet), but I see no objection as long as the functions' names (that are subject to being changed) aren't part of the already publicly released API. Otherwise, a new _major_ release may be in order when addressing this issue.

commit 7d53f98d (2023-12-15): Add snake case method alternatives

Fixes #4
closed 2023-12-21 by sfe-SparkFro