sparkfun/Qwiic_I2C_Py #11: Specify I2C bus

Cale, 2023-05-23:
I'm wondering if there's a way to specify the I2C bus that I want to work with on a system with multiple buses?

I'm working with a Libre Computer board that has multiple I2C buses. Unfortunately, the Qwiic library seems to only work with the first bus in the list. (/dev/i2c-0 vs /dev/i2c-1) Of course, my LCD is attached to the second bus in the list. Thanks.

santaimpersonator, 2023-05-23:
* I'm transferring this over to the appropriate repository.
* Since this isssue is pertaining to hardware that we currently don't [support](https://github.com/sparkfun/Qwiic_I2C_Py#supported-platforms), I will label this issue as `invalid` for now.

---
### For tracking purposes:
Feel free to add a 👍 reaction on this comment, if you need this support. Would the ~50th person ping us again, so our team can revist adding support for other hardware.

Potential fix *(untested)*:
``` py
def _connectToI2CBus(iBus = 1):

	try:
		import smbus2
	except Exception as ee:
		print("Error: Unable to load smbus module. Unable to continue", file=sys.stderr)
		return None

	daBus = None

	error=False

	# Connect - catch errors 

	try:
		daBus =  smbus2.SMBus(iBus)
```

santaimpersonator, 2023-05-23:
Upon further review of the code, users can use the set/get attribution or manually declare the bus. Closing issue; please feel free to reopen.

``` py
In [1]: import qwiic
In [2]: bus = qwiic.qwiic_i2c.getI2CDriver()
In [13]: import smbus2
In [16]: bus._i2cbus = smbus2.SMBus(1)
In [18]: bus.__setattr__("_i2cbus",smbus2.SMBus(1))
```

closed 2023-05-23 by santaimpersonator