sparkfun/Qwiic_I2C_Py #12: Unable to load I2C driver for this platform

ensarkarabudak, 2023-08-08:
Hi,

I can't get this driver to work with my **Raspberry Pi Pico W** (RP2040). I get an error when I connect any of Sparfun's qwiic sensors, despite installing all dependent packages etc.

Here are some examples of sensors:

[0.91 in. OLED Display](https://github.com/sparkfun/Qwiic_OLED_Display_Py)
```
SparkFun Qwiic OLED Display - Hello Example

Unable to load I2C driver for this platform.
Unable to load the I2C driver for this device
The Qwiic OLED Display isn't connected to the system. Please check your connection
```

but it shows up on the I2C bus:

```
Scan i2c bus...
i2c devices found: 1
Decimal address:  60  | Hexa address:  0x3c
```

[Qwiic TMP102 Sensor](https://github.com/sparkfun/Qwiic_TMP102_Py)

```
SparkFun Qwiic TMP102 Sensor Test Example

Unable to load I2C driver for this platform.
Unable to load the I2C driver for this device
Initialized.
Traceback (most recent call last):
  File "<stdin>", line 27, in <module>
  File "<stdin>", line 21, in runExample
  File "/lib/qwiic_tmp102.py", line 149, in read_temp_f
  File "/lib/qwiic_tmp102.py", line 122, in read_temp_c
AttributeError: 'NoneType' object has no attribute 'readBlock'
```

```
Scan i2c bus...
i2c devices found: 1
Decimal address:  96  | Hexa address:  0x60
```

I get this error on every qwiic sensor. Does this library not support it, how can I make it work, if someone can help me I would greatly appreciate it.
By the way, I saw such a PR: [add micropython RP2040 i2c driver ](https://github.com/sparkfun/Qwiic_I2C_Py/pull/3)

No RP2040 support?

ensarkarabudak, 2023-08-08:
Raspberry Pi is on the list but can you confirm if this includes Raspberry Pi Pico W?

santaimpersonator, 2023-08-08:
To clarify... Raspberry Pi single board computers; not the RP2040 microcontroller

ensarkarabudak, 2023-08-08:
Thank you for your interest. So how is it possible for me to run Qwiic sensors with the Pico W?

santaimpersonator, 2023-08-08:
You'd have to port the existing Python packages to MicroPython and then adapt them to use a compatible I2C driver/package for MicroPython.

*\*I'm going to label this issue as `invalid` because it doesn't pertain to a supported platform. Please, close out this issue when you are done.*

closed 2023-08-09 by ensarkarabudak

santaimpersonator, 2023-12-14:
@ensarkarabudak Turns out we needed this for a new project... 
 We are starting to adapt the code to work with MicroPython on the RP2040: https://github.com/sparkfun/Qwiic_I2C_Py/commit/bac6b94c7a758b05a85f1c96ec934d2828a52978

ensarkarabudak, 2023-12-14:
This is great news! 
I have many sensors, and if this support comes, I have many project ideas. I will make these projects and publish them. Thanks in advance! I'm looking forward to it, please keep me informed! @santaimpersonator

santaimpersonator, 2023-12-14:
No problem. I just talked to the engineer on the project, he said that everything for MircoPython should mostly be working with a few caveats *(i.e. Qwiic_Py package isn't working yet, pins are hard coded, haven't tested all our packages yet, etc.)*

* This is where you can [modify the I2C pins](https://github.com/sparkfun/Qwiic_I2C_Py/blob/master/qwiic_i2c/micropython_i2c.py#L48) for the RP2040; we currently have it hard coded for the [XRP platform](https://www.sparkfun.com/products/22230):
* If you run into any of our packages that don't work or have a package you want included, create a new issue and tag @sfe-SparkFro

ensarkarabudak, 2023-12-14:
Understood, thanks. It's been a long time since I tested. Last time, I could hardly make any sensors work. By the way, I'm using [Pico LTE](https://sixfab.com/product/sixfab-pico-lte/), which has a Qwiic port. I connect sensors through Qwiic. An example project of mine: [Air Quality Monitoring Station using Pico LTE and Thingspeak](https://www.hackster.io/ensarkarabudak/air-quality-monitoring-station-using-pico-lte-and-thingspeak-ed5b8c)

santaimpersonator, 2023-12-14:
Oooh, that is a nifty LTE board!

With regards to the air quality sensor, we haven't created a Python package for the ENS160 sensor. With the shortage of Raspberry Pi boards during the pandemic, our development of Python packages fell behind with the drop in sales. I just created an issue to add Python/MicroPython support for that sensor

ensarkarabudak, 2023-12-15:
Thank you!

Yes, I even have all the sensors below, but I am having difficulties because they do not have support. If there is support from the libraries, many users can easily use these sensors.

SparkFun Atmospheric Sensor Breakout - BME280 (Qwiic)
SparkFun Air Quality Breakout - CCS811
SparkFun 6 Degrees of Freedom Breakout - LSM6DSO (Qwiic)
SparkFun GPS Breakout - Chip Antenna, SAM-M10Q
SparkFun Humidity Sensor Breakout - SHTC3 (Qwiic)
SparkFun Qwiic PIR - 170uA (EKMC4607112K)
SparkFun Qwiic Led Stick APA102C
SparkFun Qwiic Soil Moisture Sensor
SparkFun Qwiic Triple Axis Accelerometer Breakout BMA400
SparkFun Ambient Light Sensor VEML6040 Qwiic
SparkFun STTS22H Qwiic Temp. Sensor

Also, if you need the Pico LTE card during the development process, feel free to let me know :)

sfe-SparkFro, 2023-12-15:
Hi there!

As mentioned above, I've been going through a selection of our Qwiic device Python drivers and validating them in MicroPython. I haven't yet touched most of the ones you've listed (we have so many Qwiic products...), but the BME280 and Soil Moisture Sensor should be working in MicroPython. Are there specific issues you're having with them? If so, please submit an issue in the respective repos and I'll take a look when I get the chance!

https://github.com/sparkfun/Qwiic_BME280_Py
https://github.com/sparkfun/Qwiic_Soil_Moisture_Sensor_Py

Thanks!