sparkfun/Qwiic_I2C_Py #17: `qwiic_i2c.i2cDriver.scan()` doesn't detect devices.

joooeey, 2024-01-22:
I have an I2C bus. The master is a Raspberry Pi 4. There are 10 slaves, 7 of which have the same address because it's the same sensor model and I haven't set their addresses yet.

Running `i2cdetect -y 1` in the console lists 4 device addresses, as expected.

However, running `qwiic_i2c.i2cDriver.scan()` in Python returns `None`.

What could be the reason?

sfe-SparkFro, 2024-01-22:
Hi there!

Could you please try the following in Python:

`qwiic_i2c.get_i2c_driver().scan()`

Please let me know how that goes!

joooeey, 2024-01-23:
`qwiic_i2c.getI2CDriver().scan()` works as expected and shows all 10 slaves (Now all 10 because I've given them different addresses in the meantime).

I suggest to update the README to not use the undefined name `i2cDriver`.

sfe-SparkFro, 2024-01-26:
Good point! Just updated it with the correct way to get the I2C bus, plus showing how to specify which bus to use and all the read/write methods. Thanks for pointing that out, should be good now!

closed 2024-01-26 by sfe-SparkFro