sparkfun/Qwiic_I2C_Py #20: environment is externally managed

under-score, 2024-09-20:
Worked flawlessly in the past but with OS Bookworm I get

```
admin@rasp5:~ $ sudo pip install sparkfun-qwiic-i2c
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

"systemwide" did not work either", "break system" broke it and don't want a virtual environment. 
Any other ideas?

santaimpersonator, 2024-09-20:
Per the [Raspberry Pi documentation](https://www.raspberrypi.com/documentation/computers/os.html#use-python-on-a-raspberry-pi), you will need to setup a virtual environment to install Python packages with `pip`.

Otherwise, you'll need to refer to the [`pip`/PyPI documentation](https://pip.pypa.io/en/stable/). I believe there was a way to force an external installation with one of their [option flags](https://pip.pypa.io/en/stable/cli/pip_install/#options), but I never tested the functionality with it.

under-score, 2024-09-21:
thank you - yes, there is the pip option
`--break-system-packages`
while I downgraded now to the legacy version [https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy) which solved the problem.

closed 2024-09-21 by under-score