sparkfun/Qwiic_I2C_Py #26: Circuitpython test for connection problem

RetiredWizard, 2025-03-30:
The `isDeviceConnected(self, devAddress)` method in the CircuitPythonI2C class attempts to write a 0 length bytearray to the I2C device to determine if it's connected. I haven't done exhaustive testing but I have found that on the devices I have tested, attempting to write `bytearray()` to the device throws `OSError: [Errno 5] Input/output error` which causes the method to indicate that device is NOT connected.

On the other hand attempting to write `bytearray(1)` does complete without an error. I don't know if writing a single null byte to the I2C device is a safe way to test for connection though.

malcolm-sparkfun, 2025-03-31:
Hi, thanks for raising this issue. Could you please let me know which development board you are running CircuitPython on, how you created your instance of the CircuitPythonI2C class, and what I2C device you are trying to interface with?

When I try exercising this on a SparkFun ProMicro RP2350 (running Adafruit CircuitPython 9.2.5 on 2025-03-19), I am unable to reproduce the issue. Here's what I saw when testing with the TMP117 device (address 0x48) before and after unplugging it from my board:

```
Use Ctrl-] or Ctrl-x to exit this shell
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 9.2.5 on 2025-03-19; SparkFun Pro Micro RP2350 with rp2350a
>>> import qwiic_i2c
>>> foo = qwiic_i2c.get_i2c_driver()
>>> foo.isDeviceConnected(0x48)
True
>>> foo.isDeviceConnected()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 2 positional arguments but 1 were given
>>> foo.isDeviceConnected(0x48)
False
>>>
```

malcolm-sparkfun, 2025-04-07:
I was able to test this on an ESP32 board running CircuitPython and I got the same results as you. It seems that writing a zero length byte array is not a valid way to check for connections in this case (ESP32 board + CircuitPy). I'll think about the best way to change this connection test. Another possibility (other than your suggestion of writing a single null byte) would be reading a single byte:

```
>>> import board
>>> import busio
>>> bus=busio.I2C(scl=board.SCL, sda=board.SDA, frequency=100000)
>>> buff = bytearray(1)
>>> bus.readfrom_into(0x48, buff)
>>> bus.readfrom_into(0x48, buff) # Errors out when I unplug board with address 0x48
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] Input/output error
```

This is similar to what we do in the linux_i2c.py driver for the Jetson Orin Nano platform.

@sfe-SparkFro any thoughts?

malcolm-sparkfun, 2025-04-08:
Closing with commit: 670c3c02b8fb1357918bc5a9d7b25805b1ad29e8. Thank you for bringing this to our attention @RetiredWizard.

Happy Hacking!

closed 2025-04-08 by malcolm-sparkfun