sparkfun/Qwiic_I2C_Py #28: Failure to initialize on production XRP hardware

oddron, 2025-04-12:
This library fails to initialize an I2C port on the production XRP controller board.

To reproduce:

1. Install Qwiic_I2C_Py to the XRP by following the instructions at https://docs.sparkfun.com/SparkFun_Optical_Tracking_Odometry_Sensor/software_setup-Python/#install-qwiic_i2c_py
2. Execute the following code on the XRP

```
import qwiic_i2c
q = qwiic_i2c.getI2CDriver()
```

Actual result when running on an XRP controller board:
```
>>> MicroPython v1.25.0-preview.beta06 on 2025-02-17; SparkFun XRP Controller with RP2350
Type "help()" for more information.
>>> import qwiic_i2c
>>> q = qwiic_i2c.getI2CDriver()
'id' argument required
error: failed to connect to i2c bus
>>> 
```

oddron, 2025-04-12:
A possible hint at the underlying problem: The example I2C code from https://docs.micropython.org/en/latest/library/machine.I2C.html produces the same error message: `'id' argument required`

```
>>> from machine import I2C
>>> i2c = I2C(freq=400000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'id' argument required
>>> 
```

sfe-SparkFro, 2025-04-13:
Oops! Looks like we need to have XRPCode install a newer firmware version (assuming you're using XRPCode). The I2C `id` value recently became optional (which the new Qwiic_I2C_Py takes advantage of to get the default I2C bus instead of having to manually specify it), but XRPCode is installing firmware from before that feature was implemented. We'll get that resolved.

For now, please update the firmware on your XRP. XRPCode will probably prompt you to "update" your firmware, which will actually downgrade you to the old version (this was part of a temporary hot fix for a different problem), so just ignore that until XRPCode actually installs a new version. You can download the latest XRP firmware [here](https://github.com/sparkfun/micropython/releases). Note that there are 2 versions; the "minimal" version is just normal MicroPython, whereas the other version includes all of our Qwiic drivers, so no need to manually install them any more. This is still in development, but we can update our instructions once it's finalized.

Hope this helps!